exelban/stats #739: Feature Request - Custom Widget Ordering in Menu Bar

torrswanson, 2021-12-05:
Currently Happening
- Changing widget style/type of a given module means the new widget gets added to the end (left most side) of the menu bar. 
- Changing the ordering is possible, but it's cumbersome because you have to remove all the widgets, then add them all back in your desired order one-by-one. Making a change to one of the module's widgets means the user has to reorder all widgets.

Desire
- As a user with OCD, I would like to set a custom order to all the widgets in the menu bar.
- As a user, and I change workflows/apps, I sometimes like to change the widget type of a module, and I do not want to reset all my widget ordering.

How? 
- In the overall settings add reordering UI to the list of modules where this order sets the order in the menu bar. See mockup.

Mockup:
[image]

Not addressed.
- The ordering of the widgets within a module.

exelban, 2021-12-06:
Hi. Thanks for the feedback.
MacOS have this feature. And you can reorder the items in the menu bar as you want.

Duplicate of #556

exelban, 2021-12-06:
Duplicate of #556

closed 2021-12-06 by exelban