exelban/stats #743: New Feature: Weather/Temperature

techx-guy, 2021-12-07:
I would love it if you could add a little weather icon with temp using an open source weather API like openweather or something. Nothing crazy just a simple temperature would be amazing! Thanks for all your hard work, this application is amazing.

exelban, 2021-12-09:
Hi. Thanks for the proposal.
I was sure that there was already a feature request like this, but no.
I don't think that it will be a good feature for this app. Because this is monitoring for mac. Also, there is a nice open-source app that shows weather in the menu bar. I'm not sure I can do it better.

closed 2021-12-09 by exelban

techx-guy, 2021-12-15:
Gotcha, totally understand. What is the name of that open-source menu bar app for the weather?

jnvillar, 2022-12-30:
this would be nice though