exelban/stats #744: [BUG] Top processes not rendering properly

P0pCat, 2021-12-08:
QuickLookUIService (com.apple.appkit.xpc.openAndSavePanelService (Finder))
Process is just blank in Top processes (by blank I mean that there is just empty space without icon and text)

exelban, 2021-12-09:
Hi. Thanks for the feedback.
If this process is not visible in the top process, how do you figure out that it's a `com.apple.appkit.xpc.openAndSavePanelService` process?

P0pCat, 2021-12-09:
by pid

exelban, 2021-12-09:
RAM or CPU module?

P0pCat, 2021-12-10:
network

P0pCat, 2021-12-10:
idk why but it goes up sometimes but it doesn't show the amount of bytes sent

exelban, 2021-12-10:
Thanks, I will take a look at this module.

exelban, 2021-12-11:
Hi. Unfortunately, I cannot reproduce this bug. 
Could I ask you to run `nettop -P -L 1 -k time,interface,state,rx_dupe,rx_ooo,re-tx,rtt_avg,rcvsize,tx_win,tc_class,tc_mgt,cc_algo,P,C,R,W,arch` when you will see this bug. Please share the only line which is missing in the Stats top processes. I use this command to fetch top processes. So if the first row in the Stats has a missing name or command, it means that the problem with parsing the first line of `nettop`.

p0pcat, 2021-12-13:
Ok I'm still waiting till this process will be up again

p0pcat, 2021-12-16:
@exelban idk I just found that I had another phantom process going on but this time its pid wasn't matching with any other on the system (I checked using bpytop). also as the last one it didn't send or receive any data (maybe it did but idk) it was just jumping to the top net process.

p0pcat, 2021-12-16:
I just saw it again on the top process and tried killing it: failed: no such process

p0pcat, 2021-12-16:
so, I tested this thing a little more and came to the conclusion that when there are no processes **at all** running, this phantom process rises to the top

exelban, 2021-12-16:
@p0pcat hi. Thanks for the debug. Does this fantom process appear at the top or bottom of the list? Or it appears between some processes? Or both?

p0pcat, 2021-12-16:
@exelban I just made the amount of displayed processes to maximum _15_ and it seems that phantom process is mostly there, I can see it because I have small amount of network traffic, but still I'm not really sure why it sometimes goes to the top and sometimes doesn't, also it usually appears when the network traffic is not constant _dropping and rising every few second_

exelban, 2021-12-16:
Could you maybe share a screenshot with these phantom processes?

p0pcat, 2021-12-16:
> Could you maybe share a screenshot with these phantom processes?
[image]

exelban, 2021-12-16:
It does not help(

exelban, 2021-12-16:
It does not help(
Could you maybe specify at that position these processes appears?
What I mean. You have a list of 15 processes. Do phantom processes appear at the bottom of the list?

p0pcat, 2021-12-16:
I think its always there but it can't get to the top because other processes are doing more traffic, and when there is nothing it goes to the top

p0pcat, 2021-12-16:
I don't think that its a big issue, it was just a bit confusing but now I understand that its some sort of bug and there is no real process.

exelban, 2021-12-16:
I saw this behavior a few times. It happens when nettop reports less than the max number of the top processes.
I will remove 0/0 when the process does not exist. And will see if you have this bug.

exelban, 2021-12-17:
Hi.
Could you try this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/7736220/Stats.dmg.zip)

Please give me know if phantom processes are still present or not.

p0pcat, 2021-12-17:
I have updated the app, for now I don't see anything strange, I will wait.

p0pcat, 2021-12-17:
version 2.6.23 didn't fix the issue, also I tried the built you sent and the blank space changed to 
"._number_,_number_,_number_"
also the first number was the pid

closed 2021-12-19 by exelban