exelban/stats #746: FR - listed Bluetooth devices missing

johans3, 2021-12-09:
**Describe the bug**
Currently I'm unable to see the list of registered Bluetooth devices in the Bluetooth menulet drop-down menu. 

**Details:**
 - Device:  MacPro 
 - macOS: 10.14.6
 - Application version: 2.6.22

exelban, 2021-12-09:
Hi. Thanks for the feedback.
Only Apple devices work fine in this module. Could you tell me what device you expect to see in this module?

johans3, 2021-12-10:
You can access and display any kind of registered Bluetooth device

```
"IOBluetooth", "Foundation"

get_devices()

on get_devices()
	set allNames to ((current application's IOBluetoothDevice's recentDevices:0)'s valueForKey:"nameOrAddress") as list
	set pairedNames to ((current application's IOBluetoothDevice's pairedDevices())'s valueForKey:"nameOrAddress") as list
	return pairedNames
end get_devices
```

Also, it would be great to have a Bluetooth symbol in the menu bar - an icon is easier to spot and would improve the current implementation. 
I asked already but seemed not necessary. Bluetooth is already in the list of available menu items of Stats, so adding a little icon can't hurt, imho

exelban, 2021-12-11:
Could you please tell me what device you expect to see in the Bluetooth module? It's an Apple device?

closed 2021-12-19 by exelban

binoysankar, 2022-04-05:
I have a "MiSuperBassWirelessHeadphone" connected and in Stats app doesn't recognise this device so that I can add the Bluetooth widget to menu bar.
If its only apple bluetooth devices then its fine. I was thinking all bluetooth devices will be detected.