exelban/stats #749: M1 chip: scheduler and speed limits (throttling)

pavopax, 2021-12-14:
**Describe the bug**
On Apple M1 Pro chip, the recently added "Scheduler limit" and "Speed limit" values show zero (0%).

This info seems unavailable on these platforms.

I think the source of these sensors in Stats is the `pmset` command. From `man pmset`:

     -g therm shows thermal conditions that affect CPU speed. **Not available on
     all platforms.**
     -g thermlog shows a log of thermal notifications that affect CPU speed. **Not
     available on all platforms.**

And indeed, this info is absent on my Terminal:

```
~ % pmset -g therm
Note: No thermal warning level has been recorded
Note: No performance warning level has been recorded
Note: No CPU power status has been recorded
```

**Does anyone know if these settings are available on M1 chips?**

**Details:**
 - Device: MacBook Pro (14-inch, 2021)
 - macOS: macOS 12.1 Monterey
 - Application version: Stats 2.6.23

pavopax, 2021-12-14:
References #540

BitesPotatoBacks, 2021-12-14:
@pavopax Yeah, it seems like `pmset` currently does not support Apple Silicon...

Nothing on my M1 as well.

exelban, 2021-12-16:
Hi. Thanks for reporting.
Oh, maybe someone knows a way to receive limits on apple silicon?
I don't have a mac with m1 and cannot check it(

brunocastello, 2021-12-17:
The info is not present on my M1 Air 13" (2020) either. I even issued the "pmset -g therm" command to confirm it.

Then how, apps like "Sensei", know that info? "Sensei" must be getting it from somewhere else.

casperes1996, 2021-12-20:
Can anyone explain to me how these values are even interpreted? Is it to be understood that under overheating conditions (and only overheating conditions?) it is not just frequency drops (speed limit?) but also the scheduler putting more of the CPU in a halt state (scheduler limit?)?
Thanks in advance. 

An alternative approach could maybe be this?
https://developer.apple.com/library/archive/documentation/Performance/Conceptual/power_efficiency_guidelines_osx/RespondToThermalStateChanges.html

However that may also be pointless if we have actual temperature readings anyway, since I'm not sure how, if at all, this exactly relates to any form of limiting or throttling, though based on the wording the serious and critical levels do include a form of "mitigation" against rising heat levels.

exelban, 2022-01-12:
Hi. I have checked the scheduler and the speed limits for apple silicon. And I found no way to get this information. I suppose that these values are hard (or even not possible) to calculate, because of CPU architecture (efficient + performance cores). I will remove these values from the popup for apple silicon. Ping me please, if someone will find a way to obtain these values on apple silicon.

closed 2022-01-16 by exelban