exelban/stats #750: The battery of bluetooth device is unchangable

kyriejoshua, 2021-12-15:
Describe the bug:
The battery of currently registered Bluetooth device like airpods pro is unchangable.
Everytime when it connected airpods pro, the battery showed 61% and never changed.

Details:
Device: MacBook Pro M1 13.3
macOS: 12.1
Application version: 2.6.23

exelban, 2021-12-16:
Hi. Thanks for the feedback.
I'm pretty sure that AirPods must be connected to the mac for battery level update. I will take a look of how it could be done from the app level.

kyriejoshua, 2022-01-05:
Hi, I update to 2.6.26 but the problem still exists.

[image]

[image]

exelban, 2022-01-05:
Hi, thanks. I will try another way.

michaelchadwick, 2022-01-05:
I have an Apple Bluetooth Magic Trackpad 2, which is currently charging, but the Bluetooth indicator does not change. Tried disabling/enabling the Bluetooth module, and restarting the app, but no dice. I'm on Catalina (10.5.7), btw.

Thanks for a great tool!

closed 2022-01-16 by exelban