exelban/stats #751: Close & Minimize buttons covering UI when using RTL language

Idomo, 2021-12-15:
**Describe the bug**
The Close & Minimize buttons are covering the UI when using RTL language (such as Hebrew):
[image]
You may add some padding to the top or detect when RTL layout is in use and add a margin when these buttons are placed.

**Details:**
 - Device: Macbook Pro 2021 M1 Pro
 - macOS: 12.1
 - Application version: 2.6.23

exelban, 2021-12-16:
Hi. Thanks for the feedback.
I will try to fix it.

exelban, 2021-12-18:
Hi.
It's strange, looks like macOS does not mark Hebrew as an RTL language. There is an API to fetch and user layout direction. And it does not detect rtl interface when the Hebrew language is selected. I have an option to run Stats with RTL enabled, and then the app looks like this:
[image]

Yes, the close button is still over the interface (in this case the module name). And this layout can be fixed because it could be detected by API. But as I say, the Hebrew is not marked as RTL by macos.

I can detect a Hebrew language and:
1. modify the app interface to RTL
2. make a small hack for the module trigger button

The second solution will be fast than the first one. But I'm ready to make the app more friendly with RTL interface. What option will be better? Does the interface on the screenshot above look ok? (yes, there are some visual bugs but they will be fixed)

Idomo, 2021-12-18:
@exelban Well, I guess because the app itself doesn’t supports Hebrew, it makes it looks wired when the system uses Hebrew.
Im not sure why the macOS API doesn’t recognize it as RTL (I know that iOS recognizes it well in my Swift app).
I can tell that a while ago WhatsApp had recognized Hebrew/RTL and changed the style accordingly to have place for the buttons, but since some update a year or two ago (not sure if macOS or WhatsApp) it starts doing the exact thing Stats doing (as you can see in the screenshot), but their support answers once a week and just says “Delete and reinstall the app” instead of actually help.

Anyway, updates of the system might cause some issues with the recognition, so for option 1 I’ll say that the fix will probably won’t last and take too much time, also it will look pretty bad to see English in RTL.
I think it would be the best if you’ll just add a little margin to the top area so the buttons will fit any language and direction.

Thank you for the investigation and the willing to help!

exelban, 2021-12-21:
Hi. I have found a little bit better solution than a hack for a few languages. I pinned the position of the window buttons. So they will be in the right position.

closed 2021-12-26 by exelban

Idomo, 2021-12-28:
Just updated, looks good.
Thanks @exelban!