exelban/stats #752: Not seeing temperature data

worikgh, 2021-12-15:
I turned on all the sensors in the end but to no avail no data on my status bar

In general use I would expect to be using CPU Proximity

[image]

worikgh, 2021-12-15:
macOS Monterey
  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro11,5
  Processor Name:	Quad-Core Intel Core i7
  Processor Speed:	2.5 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	6 MB
  Hyper-Threading Technology:	Enabled
  Memory:	16 GB
  System Firmware Version:	428.40.10.0.0
  OS Loader Version:	540.40.4~45

worikgh, 2021-12-15:
Installed via `brew`

worikgh, 2021-12-15:
Getting a temperature reading by clicking on the CPU widget.  The data is there

[image]

exelban, 2021-12-15:
Hi. Enable at least one widget.

closed 2021-12-15 by exelban

worikgh, 2021-12-19:
I have a lot of widgets enabled.  What do you mean?

exelban, 2021-12-19:
You have no widgets enabled in the Sensor module.

worikgh, 2021-12-20:
Found it.  I did not notice it was clickable.  I quirk of me not a bug methinks.

Thank you for your attention and thank you for this useful software.