exelban/stats #755: VPN traffic is double-counted

gelpenaddict, 2021-12-17:
**Describe the bug**
When using a VPN, the VPN process's bandwidth is added to the total.

However, the VPN is just the tunnel through which all other traffic is being directed. Thus, adding the VPN process's bandwidth double counts total bandwidth.

When using a VPN, the VPN process itself's traffic should either be excluded or used solely excluding all other processes.
Previously closed at https://github.com/exelban/stats/issues/736
**Details:**
 - Device: Macbook Pro 2019
 - macOS: 11.6.1
 - Application version: 2.6.23

exelban, 2021-12-19:
Hi.
I don't understand what do you expect by duplicating an issue that already has an answer.

**tl.td.** You need to change the reader type to `process based`. 

Not everyone has this bug. I'm using protonVPN and tailscale and do not have this problem. And cannot detect the doubled traffic because "when I'm using VPN, the traffic is doubled" - is not a constant. Do you have any solutions for this?

holdit, 2021-12-20:
This happens to me, even with _process based_, with the official Wireguard client.

I remember reading somewhere that they use the "Network Extension" framework, while many of the commercial VPN services don't. Maybe this has something to do with the 2x traffic thing?

Not sure if it helps, but here's a screenshot of the problem. Brave (browser) is downloading at ~7MB/s, the WireGuardNetworkExtension process is downloading at 7.5MB/s... total 14.5MB/s.

[image]

[image]

gelpenaddict, 2021-12-20:
> Hi. I don't understand what do you expect by duplicating an issue that already has an answer.

Not sure if it's a permissions issue but I did not see an answer in the linked issue, just that it is closed with no comments.

> Not everyone has this bug. I'm using protonVPN and tailscale and do not have this problem. And cannot detect the doubled traffic because "when I'm using VPN, the traffic is doubled" - is not a constant. Do you have any solutions for this?

Potential solution is to ignore traffic from commonly known VPN processes, either a blacklist or user-configurable list.

In any case, it's ok to close if you want since there has been discussion.  Just didn't see any in the other ticket.  Would be great though if the doublecounting can be addressed.

exelban, 2021-12-20:
I think I have found a solution.

gelpenaddict, 2021-12-22:
Just curious, did the other [ticket](https://github.com/exelban/stats/issues/736) mention process based?  I don't see any comments in it. Just wondering if there were any permissions issues at play.

In my case, the process is `vpnagentd`.  So the counter is adding its bandwidth to the total, and I don't quite follow the suggestion to use "process based"? I did try both with the same results but "Interface based" would've actually seemed more intuitive as the interface traffic did not change with VPN; only the process based would change if the VPN process reports total traffic.

Is the solution to count only the VPN process if it's detected?

exelban, 2021-12-23:
there were at least 2 issues where I was describe this bug. I will add an option to split the in/out going traffic if VPN is enabled.

closed 2021-12-26 by exelban

meiser79, 2021-12-27:
Hi, I can find this new setting under "Network". I use Tunnelblick (OpenVPN client) which creates a utun device.

gelpenaddict, 2021-12-27:
> Hi, I can find this new setting under "Network". I use Tunnelblick (OpenVPN client) which creates a utun device.

I don't see it after updating to 2.6.25 and restarting it; did you have to do something else to see the setting?

gelpenaddict, 2022-01-02:
@meiser79 Not a C# guy myself; is it supposed to be a user-toggleable setting?

Due to some VPN overhead, dividing by 2 is not quite accurate but close enough.

For more precision, could let the user designate a process as VPN, and then only use that for the total if present, while still listing the others in the drop-down.

meiser79, 2022-01-03:
@gelpenaddict Just a user myself, but looking at the code, @exelban looks for interface names.

`            return !scopes.filter({ $0.key.contains("tap") || $0.key.contains("tun") || $0.key.contains("ppp") || $0.key.contains("ipsec") || $0.key.contains("ipsec0")}).isEmpty`

To make it clear, I also don't have the VPN setting, this was a typo which I corrected in my previous post.

gelpenaddict, 2022-01-26:
@exelban how do we get this setting?

pyro-gx, 2024-11-14:
Looks like this change was not tested, I switched to https://github.com/gao-sun/eul instead