exelban/stats #756: Gas gauge battery indicator lowercase

davidebretti, 2021-12-19:
Hi, 
I would like to report a small issue with "gas gauge battery" indicator (MBP 14''), first letter is lowercase instead of uppercase as all the other sensor are written, in both module and preference pane view.
Thanks a lot for your fantastic work.
[image]

exelban, 2021-12-19:
Hi. Thanks, I will fix that.

davidebretti, 2021-12-19:
Umh, I still see the error in the new release, my fault in someway?
[image]

exelban, 2021-12-19:
Wait, no. It's another bug. The app does not correctly transform sensor keys to names.

exelban, 2021-12-20:
Hi. Could you please try this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/7747177/Stats.dmg.zip)

davidebretti, 2021-12-20:
Hi!
Tried, no luck 😬

exelban, 2021-12-21:
Hi. Could you please try this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/7757211/Stats.dmg.zip)

davidebretti, 2021-12-21:
seems fine!

closed 2021-12-26 by exelban