exelban/stats #757: Menubar spacing issue

davidebretti, 2021-12-19:
Hi,
I would like to report an issue with elements spacing in the menubar, if I add a consistent number of elements I get them too much spaced, therefore sometimes I cannot see all of them.
Tried to reproduce the issue in a scientific way to see if there's a pattern but couldn't find any.
Please check my screenshots, with 5 elements spacing is correct, adding another one (or also changing the layout having a graph) make spacing wrong.
Thank you for your support and your amazing work.

[image]
[image]

 - Device: Macbook Pro 14'' 2021
 - macOS: 12.1
 - Application version: 2.6.23

exelban, 2021-12-19:
Duplicate of #731

closed 2021-12-19 by exelban

davidebretti, 2021-12-19:
(thanks, sorry I didn't notice the duplicate)