exelban/stats #759: [Feature Request] Show bluetooth devices battery in battery panel

Junyi-99, 2021-12-19:
# Feature Description

First and foremost, thank you for developing such a great application! I love stats!

In recent days, I wondered if stats can show the battery of Bluetooth devices in the battery panel?

I mean, **it will be more efficient to have a glance at the battery panel** since the Bluetooth device and MacBook both have "battery." It's intuitive to integrate such information into the battery panel. The original way to check the Bluetooth battery requires two steps and is very annoying.

For example, first, you need to click the `"Control Center"` icon and click the "Bluetooth" icon in the control center. Then you have to wait for macOS `"expand and fade in"` animation for about 1 second. Meanwhile, I needed to search the battery level's location, which was very annoying because it's tough to locate the battery information in an animated panel.

exelban, 2021-12-19:
Hi. Thanks for the feedback.
What do you mean by battery panel? The app does have a Bluetooth module that can show the level of the Bluetooth device. For now, it works only with Apple devices. But I will find a time to add non-apple devices too. (this feature was in the app, but it causes to app crash).

Junyi-99, 2021-12-20:
Thank you for your reply!

Here is the battery panel I mentioned before:

[image]

That's approximately what I mean, what do you think of the idea?

Expecting your reply!

exelban, 2021-12-20:
I don't understand why you don't like the Bluetooth panel?
There is no reason to duplicate the same functions that the app already has in multiple places. Also, only MacBooks have a Battery module. iMac and Mac mini does not have a battery.

Junyi-99, 2021-12-20:
Oh, I didn't notice there was a Bluetooth panel, which looks great, so it looks like there's no need to add new content to the Battery Panel.

Thanks for the reply!

closed 2021-12-20 by Junyi-99

exelban, 2021-12-20:
Unfortunately, there is a bug in this module (#750), but I will try to fix it this week.