exelban/stats #760: Display & Battery Health

MasterWanna, 2021-12-20:
**Describe the bug**
1. The health of the battery exceeds 100%.
[image]
2. In line chart with value displayed, the symbol "%" is partially covered.
[image]

**Details:**
 - Device: MacBook Pro (16-inch, 2021)
 - macOS: macOS 12.1 Monterey
 - Application version: Stats 2.6.24

MasterWanna, 2021-12-20:
The health of the battery is changing between 100% and 101%

exelban, 2021-12-20:
Hi. Thanks for the feedback.
More than 100% in the battery health it's ok. It just means that your battery has a little bit more capacity than designed. 
Covered `%` will be fixed.

MasterWanna, 2021-12-20:
OK, thanks a lot!

closed 2021-12-26 by exelban

hamzabinamin, 2022-01-01:
I just got a new MacBook M1 pro and my battery health is showing as 98%, is that a bug? Cycle count is only 6 and I got it 5 days ago.

invshakil, 2022-01-06:
While my MacBook m1 pro settings are showing health is 100%, the app displays 96%. is it a bug?
[image]
[image]