exelban/stats #761: Battery Icon

holdit, 2021-12-20:
Green is used in this dropdown menu to show how much battery we have:

[image]

However, when the battery is fully charged, the green colour is gone. It's either white (dark mode) or black (light mode):

[image]

[image]

The other day I charged my laptop to 100% (it usually sits at ~80%) and when checking this menu I was like... "why is the battery icon white aka fully discharged?"

I'm not sure if this is a bug or intentional, but for consistency green should always be used to show how much battery we have, even when at 100%.

Other colours could be used (eg: red/yellow/green depending on the state of charge; two different colours for when it's using ac power or battery power; etc), but I'd stay away from white/black unless a "monochrome" theme is being used. The lack of colour might mean "empty" for some users.

(On a sidenote, it might be useful to have something in the menu bar that indicates when the battery is charging.)

exelban, 2021-12-20:
Hi. Thanks for the feedback.
The battery colors are consistant with the battery widget colors. And with the default macOS battery widget if I remember correctly. The green battery only indicated charging status. The battery is filled with color - the battery is full. When the battery is empty (only borders visible) - the battery is empty.

exelban, 2021-12-20:
PS: battery charging is indicated in the menu bar with bolt icon.

closed 2021-12-20 by exelban