exelban/stats #762: is:issue is:open The relationship between temperature sensors and performance effecting temperature is opaque

worikgh, 2021-12-22:
tl;dr  A readout of `kernal_task` cpu usage would be a useful statistic

The reason I want to monitor the temperature on my Mac is since "upgrading" to Monterey the machine has been over heating.

I diagnosed the problem with the machine (slow compile times.  Glacial) when I noted kernal_task taking about 400% of CPU.

https://setapp.com/how-to/kernel-task-high-cpu 

I have installed this programme `stats`, I llike it.

But I have found no relationship between the Max overheating and any of the temperature sensors.  (I can tell by kernal_task taking more of CPU and the powerconnector getting very hot, and I can cure it by stopping charhing and running on battery - charging a Li battery causes quite a bit of heat)

So what I am doing is monitoring `kernal_task` in activity monitor.  

Some priotiy listing of what sensors are more useful than others (but what does that even mean?) could be useful, or the CPU % of kernal task would be a very useful statistic

exelban, 2021-12-22:
Hi. Thanks for the feedback.
I'm not sure I understand your proposal correctly. Do you propose to show CPU usage of the `kernal_task` process in the app?

worikgh, 2021-12-23:
> Do you propose to show CPU usage of the kernal_task process in the app?

That seems perverse.  But it would be useful.  Because, even with your app, the only reliable indicator I have of temperature is `kernal_task` cpu (or putting my finger on the charging dongle)

I am monioring five temperature readouts to see if any are as informative but no joy so far.

I have activity monityor running all the time.

Ever since I "upgraded" to monterey this has been a problem.  I do a lot of compiling which is hard work for a computer pushing seven years old

exelban, 2021-12-23:
Hi.
Sorry but I don't see any reason to add this hack. If you need to monitor one specific process you can do it with Activity Monitor (or with top processes in the Stats). Or maybe in your case will be better to have a separate app for monitoring this process. But it will not be added to this app for sure. Sorry.

closed 2021-12-23 by exelban