exelban/stats #764: [Feature request] Kill switch

p0pcat, 2021-12-28:
Please make a kill or copy pid option when right clicking on the process from menubar

exelban, 2021-12-29:
Duplicate of #569 

I could say that this option (kill process) will be added. I must to make a few changes before, and cannot give any deadlines for this feature.

closed 2021-12-29 by exelban

movy, 2022-05-27:
Any plans to make this work for any process, not just locked?
There's a tiny util called MiniUsage, which honestly has been a lifesaver on MacOS, as it allows me kill ANY cpu-hungry process, and it saves hours of battery life daily... w/o this feature Stats is useless tbh, what's the point of observing Chrome Helper or Tradingview eating my resources in the background w/o being able to do anything with it? 

[image]