exelban/stats #765: The menu bar doesn't display any icon after installation.

IgnacioCarrera, 2021-12-28:
The menu bar icons just won't show up in my menu bar. 
I did restart the computer as that was the solution for someone. Interestingly, the icons appeared after restart for 1-2 seconds and disappeared. Nothing I do, like toggling on and off CPU or Memory gets to show anything. 
I even removed many of the icos from the menu bar in case they where "hiding" behind the notch.
Any ideas are greatly appreciated as I heard the product is really good.
Thanks,
Ignacio

Device: Macbook Pro Max 2021
macOS: Monterey 12.1
Application version: 2.6.25

exelban, 2021-12-29:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in `Applications -> Utilities -> Console`. Find the Crash Reports tab. There you must see a row called Stats.

IgnacioCarrera, 2021-12-29:
Unfortunately, I do not see any Stats line in the crash section of the console->  Stats.Is there any other place or way to help debug this? Thanks and regards,Ignacio

exelban, 2021-12-30:
Hi. Strange it looks like the app crashed at the start. But there must be a crash report in this case.
The app starts, and it stops after a few seconds?

IgnacioCarrera, 2021-12-30:
The up seems to be up an running ad I can configure it. 
However, the info on the menu bar only appears for 1 or 2 seconds max. and then disappears and ni matter what I do, it never comes back until I either restart the computer or logout and login again and will repeat the 1-2 seconds.  Reviewed the crash again and nothing related to Stats. 

Cheers.

exelban, 2021-12-30:
Maybe you have too many elements in the menu bar? Because I noticed that you have a new MacBook pro. And it has some problems with properly displaying the elements.

IgnacioCarrera, 2021-12-30:
No, I purposely remove many just in case the Stats one got hidden under the notch, but is not the case.

exelban, 2021-12-31:
Hmm, in this case, I have no idea why icons disappear. It looks like some system bug.
And what if have opened Stats settings, the app will stop working too? Or it will continue to work until you close the settings window?

IgnacioCarrera, 2021-12-31:
The app remains open until I close it. 
The issues are just the icons disappearing after 2 seconds of restarting the machine  or logging on. 
And no messages or anything. 
Very strange. 
Thanks for trying Serhiy, and have a great New year and a healthy and safe 2022. 
Ignacio

exelban, 2021-12-31:
Hmm, very strange. I even do not have an idea of what can go wrong here.
Could you enable only the CPU module with Mini widget?

Happy New Year to you too, thank you.

IgnacioCarrera, 2021-12-31:
I have disabled all except CPU and selected mini widget and there is no change... I quit the app and relaunched after the change just in case and nothing.

exelban, 2022-01-01:
Could you please check if Stats is still visible in the Activity Monitor when elements disappear from the menu bar?

IgnacioCarrera, 2022-01-01:
Happy New Year!

Yes, Stats is visible though not consuming any CPU (that I noticed). 

Cheers

IgnacioCarrera, 2022-01-01:
Quick update!
I install a second display (BENQ 27") and used as extended and the widgets all appeared in this monitor.  
Furthermoer, now they started to show in the laptop monito as well... and this time, some are hidden behind the notch but all show on the bigger screen.
VERY strange.

exelban, 2022-01-02:
Hi. Thanks for the feedback. I'm pretty sure that it's some bug in macOS.

closed 2022-01-02 by exelban