exelban/stats #766: Please Add Amperage for DC in - Feature Request/ Enhancement

jagdishadusumalli, 2021-12-31:
I have been using iStats for long and recently found stats and love it.
Thanks for the beautiful free solution.

i compared how i used iStats with stats and found only one metric value thats important for me is missing
and that is Amperage of DC In. (As i use/ test different chargers/ USB Type C cables with different 
Max Voltage and Amps ratings, i like to keep an eye on speed of charging. ( i know Dc In watts also tells the same though)

Please see the image attached for how my panel looks and exactly and which value i miss in stats.

[image]
.

exelban, 2021-12-31:
Hi. Thanks for the feedback.
I will try to find this sensor. Could you please share the output of `/Applications/Stats.app/Contents/Resources/smc list`?
Please post the output in the file, not the plain text.

jagdishadusumalli, 2021-12-31:
> /Applications/Stats.app/Contents/Resources/smc list
Here you go
[jags-smclist.txt](https://github.com/exelban/stats/files/7796483/jags-smclist.txt)

Also i saw the temperature sensor values are higher in stats and lower in istatsmenu. Around 5C difference many times.
Not sure which are correct values to trust.
Here's a document i created where i mapped each and every sensor from istatsmenu to equivalent sensor name in other apps like stats, fanny, powermetrics etc. I tried to replicate the panel i like...Here's the link of information file

https://docs.google.com/spreadsheets/d/1c1wyIXJyELhUxqY32laW9lek3h_WbXz-lP-eA4fe-rQ/edit?usp=sharing

closed 2022-01-09 by exelban