exelban/stats #769: CPU sensor bars on AppleSilicon (efficiency vs performance)

luckman212, 2022-01-02:
`NOT_A_BUG` 😊

- Stats 2.6.26
- M1 Mac Mini

I am wondering, with the CPU sensor set up as a per-core bar chart e.g.
[image]

Is there any way to know whether the bars represent an "efficiency" core or a "performance" core? Some way to set the ordering (e.g. performance = bars 1-4 and efficiency bars 5-8) or maybe a way to color the bars according to what their type is?

e.g. Activity Monitor shows output like this:

[image]

exelban, 2022-01-03:
Hi. Thanks for the feedback.
Yes, I think it could be done. But I have no idea how it could be done. I still have no MacBook with apple silicon(

Maybe you can help with this?

luckman212, 2022-01-03:
Thanks @exelban 

I don't know if I am up to that challenge but - I could try.

I looked for some examples, but it seems like this info may not be part of the API. Looks like [Shakshi3104/CoreWafer](https://github.com/Shakshi3104/CoreWafer/blob/main/Sources/CoreWafer/CoreTypeInformation.swift) implements it using a static list. Maybe there is a better way, I will keep looking.

In the meantime, I was looking at `/Modules/CPU/readers.swift` and saw what looks like maybe some typos?

`@L72-73:`
`showHyperthratedCores` ==> `showHyperthreadedCores` ?

`@Line 72:`
`CPU_hyperhreading` ==> `CPU_hyperthreading` ?

exelban, 2022-01-03:
Thanks. Yes, for sure there is no API for this. I think CPU position could word, need to be tested.

Yes, it's a typo. But "correct" typo. It has the same key as in the settings `/Modules/CPU/settings.swift`. 
It could be fixed, but then this option will be reset for all users. And I don't see a reason to do this.

luckman212, 2022-01-04:
@exelban Actually poking around that repo I found he actually made a whole menubar app that basically implements this:

https://github.com/Shakshi3104/Gaufre

[image]

exelban, 2022-01-04:
Ok, as I see they just say that the first 4 cores are efficient and 4 are performance (for M1). So in my app, you see the same order of cores in the graph widget. Just not colorized.

phu54321, 2022-01-10:
[image]

It's on IOReg. This value is read by xnu kernel itself.

https://github.com/apple/darwin-xnu/blob/2ff845c2e033bd0ff64b5b6aa6063a1f8f65aa32/osfmk/arm64/machine_routines.c#L1018

luckman212, 2022-01-11:
Nice find @phu54321 !

exelban, 2022-01-11:
Hi. As I say, now stats shows the CPU in the correct order: the first few graphs are the efficient cores and after the performance. I'm not sure I will add a colorization for them.

closed 2022-01-11 by exelban