exelban/stats #773: Suggestion - for personal preference order to be preserved.

krish-101, 2022-01-03:
Every time I open the application it doesn't remember the order that I prefer and I have to rearrange them by toggle on and off the cpu/gpu/ram to my prefer order - is there anyway you may be able to add that functionality so it remembers the last /personal preference based on my selection and not default to something else.?

Really like this stats app - appreciate. and thanks.

exelban, 2022-01-03:
Hi. MacOS saves the order of the items in the menu bar. But you need to order the items manually (CMD + drag).

closed 2022-01-03 by exelban

1Mark, 2022-04-01:
> @exelban there has been at least four issues for this I will raise an MR and add it to the readme.

#487
#881
#831
#783
#776
#773
#739
#685
#503
#449
#381