exelban/stats #778: GPU Usage not showing up on M1 Pro

wfrank21, 2022-01-05:
**Describe the bug**
I'm noticing that the GPU Usage is not showing up on my M1 Pro 14" MacBook Pro. Is there a workaround or fix that I'm not aware of?
[image]

**Details:**
 - Device: MacBook Pro 14" 2021
 - macOS: 12.1
 - Application version: [e.g. 2.6.26]

exelban, 2022-01-05:
Hi. Thanks for the feedback.
It looks strange. It must show "0%" at least. Looks like the widget is missing in the menu bar. Could you please try:
- enable another GPU widgets ("Label" widget must be visible for sure)
- disable another module (there are a few bugs in the new MacBook with the menu bar)
thanks.

wfrank21, 2022-01-06:
Well it is showing up now, but only on my external Monitors. I don't know why that is, maybe a bug of macOS itself? I checked: It is not hiding under the Notch.

exelban, 2022-01-06:
It's some bug in new the MacBook only. I don't know the reason for this. Also, I don't see any changes in the menu bar work in the new releases of MacOS. I hope it will be fixed with new releases of macOS. At this time I cannot do anything to fix this.

closed 2022-01-06 by exelban