exelban/stats #781: clicking on battery widget crashes app

MiConnell, 2022-01-08:
**Describe the bug**
Clicking on the battery widget kills the app. It seems to happen after clicking on it and then clicking away, and will happen with random 
battery icons - not the same one each time.
**Details:**
 - Device: MacBook Pro (16-inch, 2019)
 - macOS: 12.1
 - Application version: 2.6.26

example:
[image]

exelban, 2022-01-08:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

MiConnell, 2022-01-09:
Sure thing, here you go.

[crash_report](https://github.com/exelban/stats/files/7834573/crash_report.txt)

exelban, 2022-01-09:
I make a small fix in the latest version of Stats. Could you please check it?

MiConnell, 2022-01-10:
thanks for the quick update, looks like it still crashes after installing the new versio
[crash report here](https://github.com/exelban/stats/files/7836361/crash_report.txt)
n

exelban, 2022-01-11:
Thanks, I'm looking for a fix.

closed 2022-01-23 by exelban