exelban/stats #787: [Feature Request] Hide inactive / unconnected Bluetooth devices and option for Device Icon

andreisaikouski, 2022-01-10:
As in the the title:
1. Hide inactive /unconnected bluetooth devices from bluetooth section. E.g.: Airpod battery levers are displayed even when not in use.
2. Option for device icon next to battery level of bluetooth device as below image is not clear to user
(Image is of trackpad, keyboard, and airpods connected via bluetooth; airpods not in use)
[image]

exelban, 2022-01-11:
Hi. Thanks for the feedback.
A new [reader](https://github.com/exelban/stats/commit/6d80f9e0efc610f3f4828ca5f0690bd161daa22e) will fix the 1 problem. It will show only paired devices levels.

The second one is more complicated. Any indication of the device requires the new widget. And I don't have any time in the near future for this. Also, I want to have airpods/keyboard/mouse nice pictogram in the menu bar. And this type of pictogram requires a lot of work. I suppose I will find time for a new widget for Bluetooth devices, but it can be even the next year.

exelban, 2022-11-15:
Hi. Sorry, but I have no time for that. As I said, it requires a new widget with custom icons. Maybe in the future I will return to this. But not in the near future.

closed 2022-11-15 by exelban