exelban/stats #791: M1 Max: Temperatures seem incorrect

vollkorntomate, 2022-01-13:
**Describe the bug**
On my M1 Max MacBook Pro 16", the temperatures reported by Stats are significantly lower than those reported by competing tools such as iStats Menus or TG Pro.
I am currently having a GPU-intensive task running, which also uses some of the CPU and Stats reports temperatures around 45°C for most of the values, while iStats Menus and TG Pro both report around 55°C on the GPU and around 60°C on the CPU.

This issue may have its origins in #700. By not having and reading the correct sensors, the results are naturally wrong as well.

**Details:**
 - Device: MacBook Pro 16" (M1 Max, 24-core GPU)
 - macOS: 12.1
 - Application version: 2.6.27

exelban, 2022-01-13:
Hi. Thanks for the feedback.
Could you please say how you get CPU or GPU temperature in the Stats?

exelban, 2022-01-13:
And if you have read #700, you need to know that for now all monitoring systems already using the SMC for fetching the sensors. And Stats using only HID sensors. So it looks like you have the same problem that was described in #700.

PS: SMC reads for Apple Silicon are already in the master, will be available next week.

vollkorntomate, 2022-01-13:
> Could you please say how you get CPU or GPU temperature in the Stats?

In Stats, I don't read CPU or GPU temperatures. I actually just took one of the values as an approximate to the real temperature. I never really cared about if it was correct or not, since with my typical works, the M1 Max doesn't get that hot anyway. I only figured out today that the values were off when I downloaded the aforementioned tools to see my power draw (W).

closed 2022-01-13 by exelban