exelban/stats #794: [bug] Latest update (2.7.0) broke temperature sensors on M1 MacBook Air

chriskalos, 2022-01-16:
After updating to version 2.7.0 of Stats on M1 MacBook Air, the temperature sensors don't even show up in the menu and the temperature widget doesn't work in the title bar.

[image]
[image]

exelban, 2022-01-16:
Hi. You need to enable HID sensors in the settings in the [new version](https://github.com/exelban/stats/commit/98ee87fbf81924e672794a021afed161b0d1f5cd).

exelban, 2022-01-17:
The reasons why it's these sensors are disabled by default:
- it's available on M1 and looks like some sort of "hack". Because before m1 normally all sensors were in the SMC, and in the new M1 pro/max is again in the SMC
- these HID reader takes more CPU usage than all other modules together

But I will try to make it enabled by default only for the M1 macs.

bt61, 2022-01-22:
I just updated and noticed this immediately.  When I enable HID it shows two temperatures, but I can't tell which they are and see no way to set which display here like before.

closed 2022-01-23 by exelban

trinquity, 2024-12-08:
It's probably the same thing @exelban but after system update Sequoia 15.1.1 (24B2091) this happened again with my M4. Only few sensors are available right now. But it was okey with Sequoia 15 I was able to see CPU/GPU Sensors.
Sharing the sensors after osx 15.1.1 update
[image]