exelban/stats #795: Temperature not showing up

rbrtmrtn, 2022-01-16:
**Describe the bug**

I updated to Stats 2.7.0 and after the next time I restarted my computer, the temperature no longer shows up. It's a blank space on my menu bar. If I click the blank space, the Sensors panel does show up. Screenshot attached.

[image]

Looking at my preferences under Sensors, I see that nothing is toggled on. If I toggle on HID sensors or something under Temperature, e.g. Battery or Airport, a temperature will show up in the macOS menu bar. I have not touched these settings since I first installed Stats months ago, so I wonder if the upgrade to 2.7.0 maybe turned everything off accidentally?

[image]

I also am not sure which of these sensors to turn back on. I'd like to have my CPU temperature appear, but is that HID? Any advice would be much appreciated.

**Details:**
 - Device: MacBook Air M1
 - macOS: 12.1
 - Application version: 2.7.0

exelban, 2022-01-16:
Duplicate of #794

closed 2022-01-16 by exelban

rbrtmrtn, 2022-01-16:
@exelban ah OK... now I know 😄  Thanks!

rbrtmrtn, 2022-01-16:
@exelban just a quick suggestion... I might **pin** issue #794, as I would imagine this is affecting a lot of people, and they might come here confused and not see the closed issue(s) for CPU temp not showing up. That was a big help once I saw it, thanks for pointing me there!