exelban/stats #796: Fan control GUI shrinks when interacted with

Bz16, 2022-01-17:
**Describe the bug**
When the three "fan mode" button of either of the fan is interacted with, the **second** fan's mode buttons shrink. See the screen recording demonstrating the behavior.

**Details:**
 - Device: Macbook Pro (16-inch, 2021), with Apple M1 Pro
 - macOS: 12.2 Beta (21D5025f)
 - Application version: 2.7.0

Unrelated question: what does the "chill" mode with the snowflake as the button do? Is that basically a "manual fan 100% speed" shortcut?

https://user-images.githubusercontent.com/32656529/149683433-eb38740e-90f9-48c5-959a-ca9c2f1b649c.mp4

Bz16, 2022-01-17:
Hmm, seems like Github didn't like my video file. Here's a Imgur link instead. https://imgur.com/a/lOPSrUD

exelban, 2022-01-17:
Hi. Thanks for the feedback.
Yeap, I have noticed this bug too.

closed 2022-01-23 by exelban