exelban/stats #798: Process item may be duplicated when item is selected

reekystive, 2022-01-17:
When I selected the process `Stats` in CPU details list, another `Stats` process may appear in the list:

[image]

But actually there is only one `Stats` process:

[image]

**Details:**

 - Device: MacBook Air (M1)
 - macOS: 12.1
 - Application version: 2.7.0

exelban, 2022-01-17:
Yes, it could be duplicated. Because you have pinned the position. And in the next moment, this process could be in another position. But you still could stop the selected process. It's mostly only for this.

closed 2022-01-17 by exelban

reekystive, 2022-01-17:
If I pinned its position, it should always be there and be excluded in other items, I think maybe this would be better?

exelban, 2022-01-17:
I will need to stop the update all processes in this case. Locking the process is only for killing the selected process. Because without that you can click on the process with position 2 but at the same moment the processes list will update and you kill another process (if someone is now on position 2).