exelban/stats #799: Sensor popup - layout constraint violation

jonsummers, 2022-01-18:
Stats v2.7.0 release
macOS 10.13.6, macOS 10.14.6, macOS 11.6.2 (M1)
Regression: new behaviour from v2.6.27
Related: #796

On setting the NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints flag (see below) 
the purple window of layout constraints violation appears for the Sensors popup window.

NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints sets a visual debugging mode of NSView constraints.
Set: (in Terminal)
defaults write eu.exelban.Stats NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints 1
Delete : 
defaults delete eu.exelban.Stats NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints
Check state:
defaults read eu.exelban.Stats
Note Stats.app needs to be quit and relaunched each time flag modified

exelban, 2022-01-18:
Hi, thanks for the feedback. But could you please explain that?

jonsummers, 2022-01-19:
AutoLayout constraint errors are shown by a purple boundary around the window,  a handy visual debug tool to debug layout constraint errors in AutoLayout.

While researching the cause of issue #716 "10.13.6 checkbox issues", changing the width of the checkbox items, caused layout constraint errors, shown in the terminal panel of Xcode running debug Stats.

Read Apple Auto Layout Guide (google this) to understand Visual Format Language used in the terminal debug output.

Debug output suggested setting NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints flag.
Xcode launch parameters in non-release debug version could set this flag and then show auto layout problems quickly rather than relying on subtle visual changes.
This flag should not be used by users. I leave it set to 1 to help debug Stats.

jonsummers, 2022-01-19:
In regards to the Sensor popup. I made a video showing how the fan control section gets smaller and smaller, and the window top jumps up and down when clicking between Automatic and Manual.

https://user-images.githubusercontent.com/93565491/150235560-7b0fb909-5510-4208-b985-8cb4a3c070e2.mp4

closed 2022-01-20 by exelban