exelban/stats #801: Option to export the configurations

Hassan0x00A, 2022-01-18:
An option to export and import the configurations would be nice. Or even an iCloud sync.

exelban, 2022-01-18:
Hi. Thanks for the feedback. 
I was thinking about this for a long time. But I'm not sure it's needed. For example, I have a different configuration of modules and widgets on all of my macs. I see this feature only for configuration backup.

exelban, 2022-01-21:
@hogg101 Hi. I suppose it's possible to share the configurations via copying the `defaults`.

exelban, 2022-01-24:
Hi. I can add an option to export the configuration. But there is no option for importing them. I'm sure that I will not find the time for implementing this. Because, as I say, I don't really see the usage of this feature. 

And for users who need to export/import Stats settings, I would recommend doing it throw `defaults`. All Stats settings are in the file `~/Library/Preferences/eu.exelban.Stats.plist`. I suppose you can share the settings by sharing this file.

closed 2022-01-24 by exelban

williamboman, 2022-01-29:
> Because, as I say, I don't really see the usage of this feature.

I think being able to version control your settings would be pretty nice. Also, for automation and recovery purposes. Anyways, thx for all your work, I really like it!

0xMetas, 2022-10-08:
It seems like all config is done through [Kit/Plugins/Store.swift](https://github.com/exelban/stats/blob/ebc548996c1c476b01f6b2f59ea4ce8888da020e/Kit/plugins/Store.swift). 

@exelban Would you be open to a PR that adds logic to use a config file (~/.stats/config.toml) if it exists?

jody-frankowski, 2023-01-13:
For people looking for an example, here is mine, using `defaults`:
https://github.com/jody-frankowski/dotfiles/blob/179956774ca939b974dd839d17b202b09ecc9578/install.sh#L151

Drallas, 2023-01-25:
Adding to this issue

Exporting my settings:
`read eu.exelban.Stats > ~/Documents/stat.json`

Importing them:
`defaults import eu.exelban.Stats ~/Documents/stat.json`

Implemented this in an Ansible Playbook (not tested yet 🧐)

```yaml
- name: Import Stats App settings from stats.json
  command: defaults import eu.exelban.Stats files/stats.json
  args:
    executable: /bin/bash
  register: import_stats_result
```

Agreeing with @0xMetas that an .toml settings file would be very nice!