exelban/stats #804: RAM usage not accurate when compared to activity monitor

kryl0, 2022-01-20:
Recently switched to an M1 MacBook Pro 13 from a 2012 MacBook Pro 13. My older MacBook also had stats installed and displayed accurate RAM usage when compared to activity monitor. Ever since I've had it on the M1, its been displaying inaccurate values by about .50 - .60.
[image]

closed 2022-01-20 by exelban

kryl0, 2022-01-20:
> Hi. Please use search in issues, there is a description for this.

Has this been solved? I didn't seem to have this issue on my older MacBook. May I have a link to the solved forum?

AB-boi, 2026-02-18:
@kryl0 I have widely differing ram stats in the stats MacOS widgets. I opened an issue too but it was closed saying it was duplicate. I searched but found no relevant issue open. I figured it might be closed. Have you found a fix/cause for this?

kryl0, 2026-02-19:
@AB-boi I've since moved to an M4 Pro and it somehow has gotten worse. I'm seeing differences of 3.4 gb of ram at most. Still no fix