exelban/stats #805: [Bug] Bluetooth device battery levels not updating

andreisaikouski, 2022-01-20:
**Describe the bug**
Bluetooth device battery levels not updating.

e.g. with Air Pods connected, and battery levels displayed, air pods chime with low battery after some use. 
Bluetooth device battery levels are still showing 66%

**Details:**
 - Device: Macbook Pro 15-inch 2019
 - macOS: 12.1 Monterey
 - Application version: 2.7.0

exelban, 2022-01-20:
Hi. Does the air pods were paired and connected to the mac?

andreisaikouski, 2022-01-20:
> Hi. Does the air pods were paired and connected to the mac?

Yes they were. Audio was coming from the Mac in question

exelban, 2022-01-21:
Hi. 
Uh, I have a little bit enough with this air pods battery levels. The problem is in the way of how these levels could be obtained. For all devices battery levels are fetching right from the device via Bluetooth protocol. But it does not work for air pods. Air pods level could be received only from the stored data in the macOS. And I have no way to update the value that is stored there. I have 2 ways: remove the air pods levels from the app or try to guess when the battery level was updated last time and purge them if it was not updated for some time. 
If someone has an idea of how it can be fixed please tell me. Because I tried almost everything that I know.

closed 2022-01-23 by exelban