exelban/stats #806: Suggestion: battery information

rcknr, 2022-01-21:
After replacing a battery I am keeping an eye on its vitals and I'm really missing the maximum capacity displayed in the battery popup. Since it is used to calculate battery health it would be great to see it and the design capacity as well.
Speaking of battery health, I think it rounded up so even 99,9% displays as 99%. Would be nice to address as well.

exelban, 2022-01-21:
Hi. Thanks for the feedback.
Yeap, capacity could be added to the popup window. And I will take a look at how it rounds the value.

rcknr, 2022-01-21:
Thanks! It appears that rounding happens when a float value is assigned to the integer variable:
https://github.com/exelban/stats/blob/212881ec617ca0fc74724967561ce1022d1968aa/Modules/Battery/readers.swift#L89

closed 2022-01-23 by exelban

rcknr, 2022-01-23:
Thanks a lot!