exelban/stats #808: Crash When Running In Background

mikemike396, 2022-01-22:
**Describe the bug**
Have had the app crash 2 times now while my machine is running. It seems to happen when I turn off my monitors and leave the computer running for a few hours. Logs attached. 

**Details:**
 - Device: M1 Mac Mini
 - macOS: 11.6.2
 - Application version: 2.7.0 (339)

[Stats_2022-01-20-005408_Michaels-Mac-mini.txt](https://github.com/exelban/stats/files/7917478/Stats_2022-01-20-005408_Michaels-Mac-mini.txt)
[Stats_2022-01-20-172243_Michaels-Mac-mini.txt](https://github.com/exelban/stats/files/7917479/Stats_2022-01-20-172243_Michaels-Mac-mini.txt)

exelban, 2022-01-22:
Hi. Thanks for the feedback. Will try to fix it.

exelban, 2022-01-23:
Hi. I have added a fix. But I'm not sure that it will help. Please give me feedback if it works or not.

closed 2022-01-23 by exelban