exelban/stats #812: Missing average temperature

ivasonn, 2022-01-24:
Missing average temperature

**Details:**
 - Device: Macbook Pro 14, 2021
 - macOS: 12.1
 - Application version: 2.7.1

[image]
[image]
[image]
[image]

exelban, 2022-01-24:
Very strange issue. What average temperature do you expect to see in the list? Why it's missing? And why must I guess that?

ivasonn, 2022-01-24:
Is this issue https://github.com/exelban/stats/issues/610 not about average temperature for CPU? sorry, my english is not good

exelban, 2022-01-24:
Yes, but it was only for HID sensors.

ivasonn, 2022-01-24:
thank a lot for answer

closed 2022-01-24 by ivasonn