exelban/stats #815: Make it compatible with RadeonSensor

elieserme, 2022-01-27:
**Describe the bug**
When using RadeonSensor (https://github.com/aluveitie/RadeonSensor) with VirtualSMC, stats does not show the GPU temp (in my case RX6600XT).

Can you make it compatible?

**Details:**
 - macOS: Monterey
 - Application version: 2.7.1

exelban, 2022-01-27:
Hi. Thanks for the feedback.
I'm not sure I can add this. First of all, I will need a permit to use this kext. Second I will need time to implement this (I'm not sure I will have time). And the last. I have no AMD card (f* miners) and cannot test it.

exelban, 2022-01-27:
Unfortunately, it's not a native kext, and this feature could use a very small amount of users. Also as I see it requires lilu and virtualSMC, to it's mostly for hackintosh, not macs.

elieserme, 2022-01-28:
Hello, 
After some research, we will go to iStat Menus because they support macs and hackintoshes, and already works well with RadeonSensor too.
Thanks for the attention and your time!

closed 2022-01-28 by exelban