exelban/stats #817: Menu bar item spacing is inconsistent

Guistac, 2022-01-29:
**Bug Description**

When adding or removing widgets, the spacing between all menu bar items toggles between regular spacing, and stretched out.
Quitting Stats will reset the menu bar to the correct spacing, restarting it will cause the issue to reappear.
This seems to be related to the notch cutout in the new MacBook Pro, since the stretching makes the menu bar items fill out the exact space on the right of the notch. (The cutout is not visible on the screenshots, but they were captured starting at the exact right end of the cutout)

> Regular menu bar item spacing
[image]

> Stretched spacing after adding a network widget
[image]

**Details:**
 - Device: M1 Pro MacBook Pro 16"
 - macOS: Bug encountered on macOS 12.1 to 12.2
 - Application version: 2.7.1

exelban, 2022-01-29:
Yeap, it's how macos works.

closed 2022-01-29 by exelban

Guistac, 2022-01-29:
That's weird, I've never experienced this before. Any suggestion on how this stretching can be avoided ?

exelban, 2022-01-29:
Nope, it works in this way only on new MacBooks. I don't find a way to disable it.

Guistac, 2022-01-29:
Hm, I guess its an os bug then.
Thanks for your time !

exelban, 2022-01-29:
Looks like macOS trying to fit all elements in the menu bar to available space starting from some width.
It's a little bit annoying, but I can't do anything with this(