exelban/stats #819: Disk speed widget is not shown in the menu bar

e-orz, 2022-02-02:
Hi,
When I choose the speed disk speed widget nothing is shown in the menu bar. I played a bit with it and if I configure the Units to not be shown then I can see numbers.
 - Device: Macbook Pro 2021
 - macOS: 12.1
 - Application version: 2.7.2

Thanks!

I am attaching screenshots.
Disk configuration:
[image]

Menu bar:
[image]

e-orz, 2022-02-02:
Strangely, it started working now. I didn't do anything but it might started working after I did some copying from some network drive.

Anyway, should I close this issue?

exelban, 2022-02-02:
Hi. Thanks for the feedback.
The widget will show only after the first data from the reader will be available. In this case disk activity reader. It's active only when there will be some i/o operations. So there is no bug. But still, thanks for the reporting.

closed 2022-02-02 by exelban