exelban/stats #823: Time Machine disk usage incorrect

Hassan0x00A, 2022-02-04:
**Describe the bug**
Time Machine disk is shown full when it isn't.

**Details:**
 - Device: MacBook Pro 2018
 - macOS: 12.2
 - Application version: 2.7.2

[image]

exelban, 2022-02-07:
Hi. Thanks for the feedback.
Yeap, it happens with some disks. macOS does not return the correct value or does not return the spacing info at all. I can build a debug version to log what exactly the system returns. Give me know if you would like to help.

Hassan0x00A, 2022-02-08:
Yeah, sure. I can help.
Could it be because the disk is encrypted? 🤔

exelban, 2022-02-09:
Yeap, it could be a reason why free space is not correct or available.

exelban, 2022-02-13:
Without further debugging, I cannot help with that.

closed 2022-02-13 by exelban