exelban/stats #828: Version 2.7.3 doesn't getting updated

Idomo, 2022-02-07:
**Describe the bug**
The update popup jumped (the app checks for updates once a day) and asked to download 2.7.3, I've clicked Download then Install after it finished with 100%.
After that nothing happen, it just stuck on the popup with 100% and the Install button did nothing.
Closing and reopening the app didn't helped, it didn't update and still doesn't let me install the update (that I hope is overriding on the old one, otherwise I have about 4 updates cached somewhere).

**Details:**
 - Device: Macbook Pro 16" M1 Pro
 - macOS: 12.1
 - Application version: 2.7.2

exelban, 2022-02-07:
Hi. Thanks for the reporting.
Yes. Unfortunately, there is a bug. I will fix the hot-fix in a few hours. You can update the app manually or switch to the `Silent` or `At start` update check. And re-run the app.

exelban, 2022-02-07:
The hotfix has already been released, but I will leave this issue open for a week.

Idomo, 2022-02-07:
Change it to `Silent` and re-open the app solved it, thanks!

exelban, 2022-02-07:
Thanks for the feedback)

closed 2022-02-14 by exelban