exelban/stats #829: Battery/GPU Indicator won't show up on the menu bar

carstenflokstra, 2022-02-07:
**Describe the bug**
When enabling the battery/gpu indicator in Stats, it does not show up in my macOS Menu bar. Only the CPU/RAM is displayed.

**Details:**
 - Device: Macbook Pro 16" - Late 2019
 - macOS: 12.2
 - Application version: 2.7.3

exelban, 2022-02-07:
Hi. Thanks for the feedback.
Does it not appear even after some time?

carstenflokstra, 2022-02-08:
Hi! It does not appear at all after some updates of macOS.

exelban, 2022-02-08:
It's very strange. I can build a debug version with logger enabled. Give me know if you would like to help with debugging.

carstenflokstra, 2022-02-08:
I have experience in Software Development. So maybe I can checkout the branch and try to run it locally on my system. Is there any guide how to setup a development environment?

exelban, 2022-02-08:
Nope, you can simply run the project in the Xcode. You will need to switch signing team to none, and a certificate to `run locally`. Put some print at `infoCallback` in the Battery or GPU module, this function must be called when the reader receives some data. Give to know if this function is triggering.

carstenflokstra, 2022-02-08:
I'll try that tonight :)

carstenflokstra, 2022-02-08:
So I debugged some. It seems like all the assets are correctly loaded. But the menu bar does not appear in the menu bar. It is not only occuring with the GPU/Battery icon. It seems like something goes wrong when I toggle (I can't see the CPU/RAM either now).  I think it is something with this if-statement in (Widget:247), because it is skipped on my machine:

```swift
if let item = self.menuBarItem, !item.isVisible {
                self.menuBarItem?.isVisible = true
            }
```            

It does not reach the `self.menuBarItem?.isVisible = true` statement which could be the reason why it is not displayed.

carstenflokstra, 2022-02-08:
Something else to mention. After restarting my Macbook, the icons are visible for an small amount of seconds. After that, the icons are gone. Icons are placed around the place where the "notch" will be on the newer models. Maybe Apple has updated the behavior of the menubar items in their newest version of macOS.

exelban, 2022-02-09:
Oh. I see. Try to disable all modules except battery/gpu. And check if they appear in the menu bar.

closed 2022-02-13 by exelban