exelban/stats #830: [Feature request] Allow RAM indicator to show only free or used memory

johnny-the-coder, 2022-02-08:
First thing is that I like the application a lot :)

It would be great though if RAM indicator could only show one of USED/FREE (in GB) memory in the menu bar.
Currently it's only possible to display both of then and because of that the font is very tiny and hard to read.

BR,
Johnny

exelban, 2022-02-13:
Hi. Thanks for the feedback.
I was thinking about this feature, and don't really see a sense to have this. If you want to monitor free/usage space it's better to monitor in pairs. If you want to see the overall memory, the percentage is much better for this.

closed 2022-02-13 by exelban

johnny-the-coder, 2022-02-17:
Hi,

For me personally it's better to see only used memory in GB. I don't want to have labels in the menu bar and if i use percentage it's configuring with percentage of cpu.

So, for me it would be very useful if i can get in menu bar, something like this: 45%, 12.4 GB.
Than is clearly visible (currently if used and free is display the font is so tiny that i can not see anything).

I guess it should not be to complicated - probably it will be reuse of current code or simple copy&paste.

BR,
Johnny

exelban, 2022-02-17:
Hi. Nope, I must create a new widget and behavior. I cannot reuse the existing widget, the Mini widget for example. Because it was designed to fit only 0-100%, and to work with this scale. But I will review my decision after will back from vacation and maybe I will decide to add a new widget.

Just a tip: if you think that something could be just copy&paste, it is for sure not like that)

johnny-the-coder, 2022-03-18:
Ok, thanks for the info, just let me know if you decided to add this feature.

johnny-the-coder, 2022-10-24:
Hi, any update on this? ))