exelban/stats #831: [Feature request] Ordering of the icons in menu bar

johnny-the-coder, 2022-02-08:
It would be great if would be possible to define the order of the icons in menu bar, currently it seems to be random.

I would like to have then always in following order (be able to configure this in the app settings):
 - ram,
 - cpu,
 - network,
 - sensors

BR,
Johnny

closed 2022-02-08 by exelban

johnny-the-coder, 2022-02-08:
Hi, by closing this issue, you would like to say, that's:
 - it's not possible,
 - it will not be implemented,
 - it's a stupid idea ;)
?

Or maybe it's possible but i'm missing something?

exelban, 2022-02-08:
Hi. Yeap, just use search by the issues.

johnny-the-coder, 2022-02-08:
Ok, i found out that i can rearrange the icons on mac (holding ^ key) and indeed it remembers the location of the icon even after restart (i guess it was not possible on widows) - that's a nice surprise :)

1Mark, 2022-04-01:
@johnny-the-coder You mean the  ⌘ key.