exelban/stats #833: Making the status bar widgets more compact.

shhh9, 2022-02-08:
**Describe the bug**
Status bar widgets are taking up a lot of real estate. Two main issues:
1. Unnecessary spacing between categories.
2. Network widget should be designed more compactly. Should be more compact like CPU.

See attachment.

**Details:**
 - Device: [e.g. Macbook Pro 14-inch 2021]
 - macOS: [e.g. 12.2]
 - Application version: [e.g. 2.7.4]

[image]

closed 2022-02-09 by exelban

validatedev, 2022-02-13:
@exelban Why the issue is closed? The spacing also annoys me a lot.

sampl3x, 2022-02-14:
I agree, on a Macbook Pro M1 with the stupid notch space is essential :)

xareelee, 2022-02-17:
Agree.

Instead of lots of menu bar items, IMHO, it's better:

- using a neat widget to show summarized info. You can click the widget to open the full info window.
- using a window to show detailed info.

Please consider this request. :)