exelban/stats #839: Since fans have been moved to sensors there is no longer a bar chart option.

mumstead, 2022-02-14:
When fan monitoring was moved to the sensors module the ability to display them as a bar chart was removed. Currently you can only see the RPMs. Can the bar chart functionality be restored?

exelban, 2022-02-14:
Hi. Yeap, the bar chart will available only in the Fans module. I will take a look if it's possible to add in the Sensors module (only for fans).

Jethro1000, 2022-02-16:
@exelban sorry, IMHO that was a worst decision. Please bring back the fan module as separate. I am back to 2.7.2 and keeping that version. I miss my BarChart
[image]

exelban, 2022-02-16:
@Jethro1000 Hi. Nope, the fans module will not be returned. Because: 
1. The fan module was just a duplicate of the sensors module with small changes and a new feature (setting the speed). 
2. Having the fans module separate makes impossible to add a feature to set fan speed based on some sensors.
3. Did you read this thread? If not, please read before posting a comment like this.

closed 2022-03-13 by exelban