exelban/stats #840: Crashing at start

sddgit, 2022-02-15:
**Describe the bug**
Stats is crashing every time it starts. Works OK on one user login, but not on another. It _may_have started after I installed the 11.6.4 update, but can't be sure. Crash log attached. I have tried re-downloading to no avail.

[Crash.txt](https://github.com/exelban/stats/files/8066235/Crash.txt)

**Details:**
 - Device: MacBook Pro mid 2014
 - macOS: 11.6.4
 - Application version: 2.7.5

exelban, 2022-02-15:
Hi. It must work now.

closed 2022-02-15 by exelban

sddgit, 2022-02-15:
Sorry, not with you. What do you mean by “it must work now”?

exelban, 2022-02-15:
It was a server error, the Stats should work now. It means that you can run the application, and it should not crash.

sddgit, 2022-02-15:
Ah ok. I didn’t realise Stats depended on a server. What is that used for?