exelban/stats #841: [Feature Request] temperature curve with fans

mpfhb, 2022-02-15:
Hello, btw I love your Stats project.

Is it possible to implement a Temp/Fanspeed curve, to set a certain fanspeed for a temperature?

like 50°C = 70% fanspeed
70°C = 90% fanspeed and so on

ty

[image]

exelban, 2022-02-28:
Hi. Thanks for the proposal.
I'm not sure that the curve will be implemented. As I always say: it's a monitoring app, not controlling. There are a lot of nice OSS apps that allow setting a temperature based on different parameters. 

There will be an option to have different levels based on one temperature sensor. But I cannot say when it will be added.

closed 2022-02-28 by exelban

xareelee, 2022-03-05:
Could you add an option for minimum fan speed?

My MacBook Pro M1 Max runs with **0 RPM** at 40~58ºC of CPU temp with `Automatic` fan speed option.

I'd like to make the minimum fan speed run with **1500 RPM** to keep the CPU cool.

xppx99, 2022-04-13:
I also would like this very much. No need for a curve, a simple Rule-Based Fan Control would be enough. Thanks!