exelban/stats #844: [Enhancement] Network graph widget enhancements proposal

bouscram, 2022-02-18:
Hi.
I like to use the network graph, but I am disturbed by the "auto-scale" of the current implementation. Would it be possible to have a setting to use a fixed value (maybe based on the interface speed, or user selectable)? 
Another change would be to have the possibility to use a logarithmic scale for the graph (as in MenuMeters).

exelban, 2022-02-28:
Hi.
I'm not sure I understand correctly. There is one way of how the chart calling works now. It's trying to fit the values to the view. If you want to implement a more complex way to visualize the data, PRs are welcome.
I'm not sure I will find a time to do this.

exelban, 2022-02-28:
How it's possible to have a constant scale?

bouscram, 2022-03-05:
What I mean is that the max height of the curve is predefined and selectable (for example 100% solid = 1Gbps), instead of adapting itself to the max visible value. 
With the current behaviour, it's impossible be see if the trafic is high or low by just looking at the curve. On the example below, is the spike 1 Gbps or 10 Kps ? 
[image]

exelban, 2022-03-05:
Ok, so imagine that the scale will be predefined: 100% = 1Gbps. It means that I will never see any value on the chart. Because the value in the range of 1-10 Mbps will soo small to visualize.

For now, there is no predefined scale. And the 100% is the max value that was in the selected time range,

bouscram, 2022-03-05:
Clearly I know that small values won't be visible, but what I like this kind of widget for is to see abnormal values (= large traffic when I don't expect any) at a glance. Maybe several max values could be used, a the user's choice.

To cope for that, MenuMeters (which I used for years before mainly switching to Stats) has two setting, one for the range and one for the display type :
[image]
[image]
Please consider such changes for future release. I guess the 1st is very easy, but the 2nd needs changes in the way the graph is calculated.

exelban, 2022-03-07:
Hi.
I'm not sure I see a reason for this. But still, I understand that someone needs this type of scaling.
I will leave this ticket open. Maybe someone would like to implement this feature.

closed 2022-08-07 by exelban

victor-marino, 2023-12-26:
Hi.

I believe this issue was closed by mistake.

The linked commit simply matches the scale for the upload and download speeds, but those scales are still set dynamically by the app instead of allowing us to set the maximum values.

I think that @bouscram was asking for (and I second this petition) is adding the option to set the scale of the network chart manually, both in the status bar and the expanded widget chart.

This option is present in most traffic monitor apps I have used, as the chart will otherwise look like this when there's only background traffic involved:
[image]

Furthermore, the scale keeps changing constantly not based on our traffic history, but on the time range that is visible in the chart at any given moment. So if we make a large download the scale will adapt to, say 1 Gbps maximum speed. But after a few seconds, when that traffic leaves the chart, the scale will suddenly go back to measure Kbps.

This is of course dependant on user preference, but as a network engineer, for me it makes a lot more sense to adjust the scale so that I just see a "flat line" when there's only background traffic. That way, the chart will only draw my attention when there's actual, significant traffic going on. This also works nicely as an "alert" whenever an app is downloading stuff in the background without our knowledge.

Given each person will have different preferences and different broadband speeds at home, I think the easiest way to accomplish this is to just add a "manual scale" option, that then allows the user to enter the desired maximum values.

Could you please consider reopening this issue?

victor-marino, 2023-12-27:
Hi again.

Even though I've never, ever developed for MacOS or used Swift, I decided to give this a try as I'd love to see this feature implemented.

Initial PR here: https://github.com/exelban/stats/pull/1746