exelban/stats #852: This app is incredible with one issue.

booooooooooooooooooo, 2022-02-21:
I love the app and everything about it but for some reason the network statistics do not work correctly. I live in a rural area so I only get a consistent 7 mbps or so internet connection as shown by the youtube statistics. However, according to the menu bar statistics It mostly reads 0 up and down with the occasional jump into the 100 to 200 kilobit range. It's not the end of the world or anything but it's just a bug I thought you should know about.  
[image]
.

Tai-Zhou, 2022-02-22:
According to my experience, your screenshot of YouTube shows the current downloading speed is 0KB (**Network Activity**, and I think the unit is even not KBps), while the **Connection Speed** only means the maximum downloading speed you can achieve. If the video is buffering (which is not continuous, as you mentioned, "occasional jump". You can also notice the "jump" of the light gray part of the progress bar), you will see colors but not black in **Network Activity**.

[image]
.

exelban, 2022-02-28:
Hi. Yeap, as @Tai-Zhou says. It's a connection speed, not the io.

closed 2022-02-28 by exelban