exelban/stats #853: Improvement proposal: switch from "Memory" to "Real Memory" for list of apps

thousandsofthem, 2022-02-22:
UI element in question: RAM->"Top processes"
[image]

Proposal: show "Real memory" instead of "Memory" in this list.
What "Activity monitor" says:

[image]

As per [this](https://apple.stackexchange.com/questions/362249/what-is-the-difference-between-memory-real-mem-and-compressed-mem) thread:

* Memory = The amount of memory the process has reserved for its use.
* Compressed mem = The amount of memory (a subset of "Memory") that the operating system has used its compression algorithm on
* Real mem = The amount of real memory that is actually used by the process (i.e. a sum of the memory that hasn't been compressed, and the size of the result of the compression algorithm)

exelban, 2022-02-28:
Hi. Nope, all memory will be shown. There is no sense to show only `Real` because compressed are taking space in the RAM too. And even if you see that some app takes 1GB `real` memory it does not mean that you have available 7GB.

closed 2022-02-28 by exelban

thousandsofthem, 2022-02-28:
Okay, thanks. My understanding though "Real" is noncompressed memory + compressed memory (size after compressing), and
"Memory" is noncompressed memory + compressed memory (size before compressing).
Just in case summed up numbers: top17 processes consume 16.5GB right now,  which is more than 16GB memory physically present on device. 
So "Memory" is not actual memory used for sure

exelban, 2022-02-28:
I don't see an API to receive this data per process. And also, it will add a lot of misunderstanding when users will see not the same values in the activity monitor and Stats.

thousandsofthem, 2022-02-28:
Well, calculating memory is complicated for sure. Basically all numbers alone are misleading in some way. I dont know good solution either. Thanks for taking your time!

1Mark, 2022-04-11:
@exelban would be nice to have a setting to switch to real if you want it

use case is this https://stackoverflow.com/questions/58308169/docker-for-mac-memory-usage-in-com-docker-hyperkit